androidthings/drivers-samples #1: apa102 example - missing loop for LED lighting?

blundell, 2016-12-15:
In the apa102 example there is no loop of the animation thread, but in pwmspeaker the example is looped.

apa102:
https://github.com/androidthings/drivers-samples/blob/cfa3c7fbf376600b990e0bf06999ffbded59144c/apa102/src/main/java/com/example/androidthings/driversamples/MainActivity.java#L63

pwmspeaker:
https://github.com/androidthings/drivers-samples/blob/cfa3c7fbf376600b990e0bf06999ffbded59144c/pwmspeaker/src/main/java/com/example/androidthings/driversamples/SpeakerActivity.java#L98

I was wondering if this is a mistake for the apa102 example. Should it be looped? (because otherwise it'll just light up the LED's once then stop right?

jdkoren, 2016-12-15:
Yes, we should either loop the `Runnable` or include instructions to restart the app like we do for the `pwmspeaker` sample.

blundell, 2016-12-15:
👍 , thought I was going crazy for a second :-)

blundell, 2016-12-17:
fixed in https://github.com/androidthings/drivers-samples/commit/438b8736481d44b3c6c3196dd5f8cb3e8737165c

closed 2016-12-17 by blundell