androidthings/drivers-samples #2: PWMServo sample not working with Raspberry Pi 3

rno, 2016-12-18:
Hello,

I can't get the [pwmservo](https://github.com/androidthings/drivers-samples/tree/master/pwmservo) sample running. The servo I'm using works properly (I tried with an Arduino Uno).

As a side note, I also tried the other [pwn sample](https://github.com/androidthings/sample-simplepio/tree/master/pwm) without success.

proppy, 2016-12-19:
@rno can you share a picture on how you wired the servo to the Raspberry Pi?

rno, 2016-12-20:
@proppy, here's the picture of my wiring:

[image]

Sorry for the cables colors not matching the ones from the servo, that the only ones left I have.

jdkoren, 2016-12-20:
@rno The wiring looks correct to me. You may have to change some code in the sample app to match the specifications of the servo; in particular, you may need to call `setPulseDurationRange(0.5, 2.5)` based on the technical details [listed here](https://www.adafruit.com/product/154). Can you give that a try and let us know if that helps?

rno, 2016-12-22:
Hi @jdkoren, it still does not work for me :-/

jdkoren, 2016-12-22:
@rno Do you have another Servo you can try? Perhaps you can also try a different PWM pin?

rno, 2016-12-23:
Ok, I just removed a servo from another project to test it and it works.

I still don't understand why the other one is not working even though it's working fine on the Arduino...

Sorry for the trouble and thanks a lot for your help!

closed 2016-12-23 by rno

proppy, 2017-01-09:
@rno can you tell us which servo library (if any) you were using w/ the UNO?

rno, 2017-01-16:
@proppy, I'm using the arduino servo library, without changing the min and max (it worked out of the box).

Sorry for replying only now.