androidthings/drivers-samples #3: GPS driver sample buffer overflow

ckoessler, 2016-12-26:
I am trying the sample for the GPS driver with a raspberry pi 3 and the Ultimate GPS V3 breakout board.

The GPS board is connected following this schematics:
[image]

When launching the sample app, I get the following error:
```
12-26 21:11:50.561 1299-1299/com.example.androidthings.driversamples E/AndroidRuntime: FATAL EXCEPTION: main
                                                                                       Process: com.example.androidthings.driversamples, PID: 1299
                                                                                       java.nio.BufferOverflowException
                                                                                           at java.nio.Buffer.nextPutIndex(Buffer.java:508)
                                                                                           at java.nio.HeapByteBuffer.put(HeapByteBuffer.java:142)
                                                                                           at com.google.android.things.contrib.driver.gps.NmeaGpsModule.processBuffer(NmeaGpsModule.java:178)
                                                                                           at com.google.android.things.contrib.driver.gps.NmeaGpsModule.readUartBuffer(NmeaGpsModule.java:160)
                                                                                           at com.google.android.things.contrib.driver.gps.NmeaGpsModule.access$000(NmeaGpsModule.java:35)
                                                                                           at com.google.android.things.contrib.driver.gps.NmeaGpsModule$1.onUartDeviceDataAvailable(NmeaGpsModule.java:139)
                                                                                           at com.google.android.things.pio.UartDevice$UartDeviceCallbackDispatch.dispatchInterruptEvent(UartDevice.java:507)
                                                                                           at com.google.android.things.pio.CallbackDispatch.onFileDescriptorEvents(CallbackDispatch.java:127)
                                                                                           at android.os.MessageQueue.dispatchEvents(MessageQueue.java:282)
                                                                                           at android.os.MessageQueue.nativePollOnce(Native Method)
                                                                                           at android.os.MessageQueue.next(MessageQueue.java:323)
                                                                                           at android.os.Looper.loop(Looper.java:136)
                                                                                           at android.app.ActivityThread.main(ActivityThread.java:6077)
                                                                                           at java.lang.reflect.Method.invoke(Native Method)
                                                                                           at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:865)
                                                                                           at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:755)
```

devunwired, 2016-12-27:
Thanks for the report, I will look into this. Can you provide me a few more details about the occurrence?

- Do you see this happen every time you launch the sample? Does it only occur on the first run?
- Is it immediately after launch or after the app has been running for a period of time?

ckoessler, 2016-12-27:
- First time I ran the sample, I got the "No permission" message. I rebooted my RBPi and now consistently get this error everytime I run the sample.
- The error happen immediately after launch.

I do see some website saying that the uart has some issue on the RBPi3:

> For the Raspberry Pi 3 you need to explicitly enable the serial port on the GPIO pins. The reason for this is a change with the Pi 3 to use the hardware serial port for Bluetooth and instead use a slightly different software serial port for the GPIO pins. A side effect of this change is that the serial port will actually change speed as the Pi CPU clock throttles up and down--this will unfortunately cause problems for most serial devices like GPS receivers!

Does that apply to the android things image?

devunwired, 2016-12-27:
Ah, so this might be part of the issue. By default, the RPi3 UART is enabled on Android Things, but it is attached to the debug console. Have you seen [this SO post](http://stackoverflow.com/questions/41127018/uart-peripherals-on-android-things-for-raspberry-pi-3) on using UART peripherals on RPi3? The console interaction may be confusing your peripheral and causing excess garbage on the serial lines that is clogging the fixed buffer in the sample.

Beyond that, there may be some things we can do in the sample to better manage the buffer when garbage data is present.

ckoessler, 2016-12-27:
I did see this post, I wasn't sure if that applied to this sample since I didn't see anything in the readme. I will try to follow the SO solution and will post an update here when done.

ckoessler, 2016-12-27:
I followed the SO answer and removed `console=serial0,115200` from `cmdline.txt` and I still have the same issue.

ckoessler, 2016-12-28:
I enabled the Logs from the contrib-driver lib and saw a new error message: `12-28 17:53:29.862 1378-1378/com.example.androidthings.driversamples W/NmeaParser: Invalid checksum (62), expected 108`

Could that be the cause of the overflow? What could be causing that?

ckoessler, 2016-12-28:
After increasing the buffer size by 4, I was able to get few messages in. I see some messages have some junk that would explain the overflow:

```
12-28 23:38:17.393 2366-2366/? D/XXX: message: ��GPGGA,233817.000,3742.1931,N,12208.3976,W,1,04,1.96,164.3,M,-25.5,M,,*58
12-28 23:38:17.394 2366-2366/? D/XXX: Buffer reset
12-28 23:38:17.394 2366-2366/? D/XXX: message: GPGSA,A,3,23,03,26,22,,,,,,,,,2.20,1.96,1.00*0B
12-28 23:38:17.395 2366-2366/? D/XXX: Buffer reset
12-28 23:38:17.544 2366-2366/com.example.androidthings.driversamples D/XXX: message: GP��������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������RMC,233817.000,A��������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������,3742.1931,N,12208.3976,W,0.42,205.67,281216,,,A*71
12-28 23:38:17.545 2366-2366/com.example.androidthings.driversamples D/XXX: Buffer reset
12-28 23:38:17.545 2366-2366/com.example.androidthings.driversamples D/XXX: message: GPVTG,205.67,T,,M,0.42,N,0.78,K,A*32
12-28 23:38:17.546 2366-2366/com.example.androidthings.driversamples D/XXX: Buffer reset
```

I have no idea where that junk could come from...

devunwired, 2016-12-29:
Hmmm, so this is the type of behavior I would expect to see if the serial debug console were still enabled on the UART. It transmits at a different baud rate, which is usually why the data comes across as garbage. Would you be able to modify the logs to determine what those non-printable characters are? I'm guessing `\x00`, but I'd like to be certain.

@proppy Have you seen this behavior on RPi? Can you think of something else that needs to be disabled?

ckoessler, 2016-12-30:
[This SO answer](http://stackoverflow.com/a/41374016/1245894) helped me get the sample to work although the uart is still getting a lot of garbage.

devunwired, 2017-01-19:
Change merged in androidthings/contrib-drivers#30 that should fix this.
We can close the bug when the driver+sample are updated.

proppy, 2017-01-25:
Fixed with `com.google.android.things.contrib:driver-gps:0.2`

closed 2017-02-23 by ckoessler

jurandirj, 2017-06-14:
Hi.
I'm using the driver with a raspberry and a 6m ublox.
It happens that I only receive lat and long. The other data are received with value = 0.
I checked all the classes that make up the library and I did not find what might be happening. Any idea what that might be?

proppy, 2017-06-15:
@jurandirj can you file a separate issue about this?

jurandirj, 2017-06-15:
Sorry, there is already a topic about this problem. I should have commented on it.

https://github.com/androidthings/drivers-samples/issues/7

danangindrak, 2018-01-18:
hi, i want to know list gps device support version. i use this driver ublox neo 6m and gps cant show my location. 
[image]

devunwired, 2018-01-18:
@danangindrak How is your module configured? From the datasheet, that GPS module's output and baud rate are variable based on the external CFG pins.

danangindrak, 2018-01-18:
@devunwired i use this
```
public static final int UART_BAUD = 9600;  //From GPS datasheet
    public static final float ACCURACY = 2.5f; // From GPS datasheet
    private static final String TAG = GpsService.class.getSimpleName();
    private static final String UART_BUS = "UART0";
    private NmeaGpsDriver mGpsDriver;
```

devunwired, 2018-01-18:
@danangindrak Sorry, I meant how have you configured the actual hardware module? The uBlox part is configurable and different modules from different vendors are set up with different defaults.