androidthings/drivers-samples #6: Issue with Bmx280SensorDriver

MacSimmy, 2017-03-25:
Hi,
I have BMP 280 sensor from Ada fruit. While running BMP 280 sample, it is crashing with following logs:
03-25 13:51:04.250 1271-1271/? E/TemperatureActivity: Error configuring sensor
                                                      com.google.android.things.pio.PioException: android.os.ServiceSpecificException: I/O error
                                                          at com.google.android.things.pio.I2cDevice.readRegByte(I2cDevice.java:104)
                                                          at com.google.android.things.contrib.driver.bmx280.Bmx280.connect(Bmx280.java:171)
                                                          at com.google.android.things.contrib.driver.bmx280.Bmx280.<init>(Bmx280.java:149)
                                                          at com.google.android.things.contrib.driver.bmx280.Bmx280SensorDriver.<init>(Bmx280SensorDriver.java:54)
                                                          at com.example.androidthings.driversamples.TemperatureActivity.onCreate(TemperatureActivity.java:62)
                                                          at android.app.Activity.performCreate(Activity.java:6662)
                                                          at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1118)
                                                          at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2599)
                                                          at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2707)
                                                          at android.app.ActivityThread.-wrap12(ActivityThread.java)
                                                          at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1460)
                                                          at android.os.Handler.dispatchMessage(Handler.java:102)
                                                          at android.os.Looper.loop(Looper.java:154)
                                                          at android.app.ActivityThread.main(ActivityThread.java:6077)
                                                          at java.lang.reflect.Method.invoke(Native Method)
                                                          at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:865)
                                                          at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:755)
                                                       Caused by: android.os.ServiceSpecificException: I/O error
                                                          at android.os.Parcel.readException(Parcel.java:1697)
                                                          at android.os.Parcel.readException(Parcel.java:1636)
                                                          at com.google.android.things.pio.IPeripheralManagerClient$Stub$Proxy.I2cReadRegByte(IPeripheralManagerClient.java:1431)
                                                          at com.google.android.things.pio.I2cDevice.readRegByte(I2cDevice.java:102)
                                                          at com.google.android.things.contrib.driver.bmx280.Bmx280.connect(Bmx280.java:171) 
                                                          at com.google.android.things.contrib.driver.bmx280.Bmx280.<init>(Bmx280.java:149) 
                                                          at com.google.android.things.contrib.driver.bmx280.Bmx280SensorDriver.<init>(Bmx280SensorDriver.java:54) 
                                                          at com.example.androidthings.driversamples.TemperatureActivity.onCreate(TemperatureActivity.java:62) 
                                                          at android.app.Activity.performCreate(Activity.java:6662) 
                                                          at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1118) 
                                                          at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2599) 
                                                          at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2707) 
                                                          at android.app.ActivityThread.-wrap12(ActivityThread.java) 
                                                          at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1460) 
                                                          at android.os.Handler.dispatchMessage(Handler.java:102) 
                                                          at android.os.Looper.loop(Looper.java:154) 
                                                          at android.app.ActivityThread.main(ActivityThread.java:6077) 
                                                          at java.lang.reflect.Method.invoke(Native Method) 
                                                          at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:865) 
                                                          at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:755) 

Can somebody please helps me here?

spitfire88, 2017-04-12:
You could check if the sensor is connected properly to your device. I noticed this behavior on the Edison platform. The device wasn't connected when I ran i2cdetect (-r BUS_NUM). If the device isn't connected i2c calls from PIO will fail.

See https://linux.die.net/man/8/i2cdetect.

blundell, 2017-07-02:
@spitfire88 I assume you didn't run this though `adb shell` then?

```
rpi3:/ $ i2cdetect 
/system/bin/sh: i2cdetect: not found
```

RichardisMeok, 2017-09-25:
I have Ginkgo USB-I2C Adapter when i run i2c sample with Resberry Pi 3, it is crashing with following logs:
09-25 12:33:48.428 2011-2011/? W/System.err: com.google.android.things.pio.PioException: android.os.ServiceSpecificException: I2C1 driver failure
09-25 12:33:48.429 2011-2011/? W/System.err:     at com.google.android.things.pio.I2cDeviceImpl.<init>(I2cDeviceImpl.java:43)
09-25 12:33:48.430 2011-2011/? W/System.err:     at com.google.android.things.pio.PeripheralManagerService.openI2cDevice(PeripheralManagerService.java:232)
09-25 12:33:48.431 2011-2011/? W/System.err:     at com.jxkj.consumerterminal.util.I2CDemo.setUp(I2CDemo.java:42)
09-25 12:33:48.431 2011-2011/? W/System.err:     at com.jxkj.consumerterminal.activity.HomeActivity.onCreate(HomeActivity.java:57)
09-25 12:33:48.431 2011-2011/? W/System.err:     at android.app.Activity.performCreate(Activity.java:6662)
09-25 12:33:48.431 2011-2011/? W/System.err:     at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1118)
09-25 12:33:48.431 2011-2011/? W/System.err:     at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2599)
09-25 12:33:48.432 2011-2011/? W/System.err:     at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2707)
09-25 12:33:48.432 2011-2011/? W/System.err:     at android.app.ActivityThread.-wrap12(ActivityThread.java)
09-25 12:33:48.432 2011-2011/? W/System.err:     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1460)
09-25 12:33:48.432 2011-2011/? W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:102)
09-25 12:33:48.433 2011-2011/? W/System.err:     at android.os.Looper.loop(Looper.java:154)
09-25 12:33:48.433 2011-2011/? W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:6077)
09-25 12:33:48.433 2011-2011/? W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
09-25 12:33:48.434 2011-2011/? W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:865)
09-25 12:33:48.434 2011-2011/? W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:755)
09-25 12:33:48.434 2011-2011/? W/System.err: Caused by: android.os.ServiceSpecificException: I2C1 driver failure
09-25 12:33:48.435 2011-2011/? W/System.err:     at android.os.Parcel.readException(Parcel.java:1697)
09-25 12:33:48.435 2011-2011/? W/System.err:     at android.os.Parcel.readException(Parcel.java:1636)
09-25 12:33:48.435 2011-2011/? W/System.err:     at com.google.android.things.pio.IPeripheralManagerClient$Stub$Proxy.OpenI2cDevice(IPeripheralManagerClient.java:1367)
09-25 12:33:48.435 2011-2011/? W/System.err:     at com.google.android.things.pio.I2cDeviceImpl.<init>(I2cDeviceImpl.java:41)
09-25 12:33:48.436 2011-2011/? W/System.err: 	... 15 more

mulima, 2017-12-17:
Same problem here. The output of the i2cdetect command is as below;

D:\Android\sdk\platform-tools>adb shell
rpi3:/ $ i2cdetect -V
/system/bin/sh: i2cdetect: not found

mangini, 2017-12-19:
@mulima 
There's no i2cdetect on Android Things. You can list the I2C pins (and other protocols) by using the [pio tool](https://developer.android.com/things/sdk/pio/pio-cli.html#list): `pio list i2c`

@RichardisMeok 
As for the Ginkgo USB-I2C, I don't believe it will work right out of the box. You might need to open your USB device using USBHost APIs, implement the Ginkgo protocol and create a [I2C Bus](https://developer.android.com/things/reference/com/google/android/things/pio/I2cBusDriver.html) to map the devices connected through USB to the I2C protocol.

closed 2017-12-19 by mangini