androidthings/drivers-samples #8: cap12xx sample doesn't work for Pico i.MX7D

markmcd, 2017-06-08:
I'm attempting to use the capacitive buttons (A/B/C), but the sample app fails to initialise:

```
06-08 06:23:26.544 4425-4425/? W/CaptouchActivity: Unable to open driver connection
                                                   com.google.android.things.pio.PioException: android.os.ServiceSpecificException: I/O error
                                                       at com.google.android.things.pio.I2cDeviceImpl.writeRegByte(I2cDeviceImpl.java:121)
                                                       at com.google.android.things.contrib.driver.cap12xx.Cap12xx.setInputsEnabled(Cap12xx.java:451)
                                                       at com.google.android.things.contrib.driver.cap12xx.Cap12xx.init(Cap12xx.java:257)
                                                       at com.google.android.things.contrib.driver.cap12xx.Cap12xx.<init>(Cap12xx.java:203)
                                                       at com.google.android.things.contrib.driver.cap12xx.Cap12xxInputDriver.<init>(Cap12xxInputDriver.java:99)
                                                       at com.google.android.things.contrib.driver.cap12xx.Cap12xxInputDriver.<init>(Cap12xxInputDriver.java:57)
                                                       at com.example.androidthings.driversamples.CaptouchActivity.onCreate(CaptouchActivity.java:51)
                                                       at android.app.Activity.performCreate(Activity.java:6662)
                                                       at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1118)
                                                       at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2599)
                                                       at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2707)
                                                       at android.app.ActivityThread.-wrap12(ActivityThread.java)
                                                       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1460)
                                                       at android.os.Handler.dispatchMessage(Handler.java:102)
                                                       at android.os.Looper.loop(Looper.java:154)
                                                       at android.app.ActivityThread.main(ActivityThread.java:6077)
                                                       at java.lang.reflect.Method.invoke(Native Method)
                                                       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:865)
                                                       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:755)
                                                    Caused by: android.os.ServiceSpecificException: I/O error
                                                       at android.os.Parcel.readException(Parcel.java:1697)
                                                       at android.os.Parcel.readException(Parcel.java:1636)
                                                       at com.google.android.things.pio.IPeripheralManagerClient$Stub$Proxy.I2cWriteRegByte(IPeripheralManagerClient.java:1508)
                                                       at com.google.android.things.pio.I2cDeviceImpl.writeRegByte(I2cDeviceImpl.java:119)
                                                       at com.google.android.things.contrib.driver.cap12xx.Cap12xx.setInputsEnabled(Cap12xx.java:451) 
                                                       at com.google.android.things.contrib.driver.cap12xx.Cap12xx.init(Cap12xx.java:257) 
                                                       at com.google.android.things.contrib.driver.cap12xx.Cap12xx.<init>(Cap12xx.java:203) 
                                                       at com.google.android.things.contrib.driver.cap12xx.Cap12xxInputDriver.<init>(Cap12xxInputDriver.java:99) 
                                                       at com.google.android.things.contrib.driver.cap12xx.Cap12xxInputDriver.<init>(Cap12xxInputDriver.java:57) 
                                                       at com.example.androidthings.driversamples.CaptouchActivity.onCreate(CaptouchActivity.java:51) 
                                                       at android.app.Activity.performCreate(Activity.java:6662) 
                                                       at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1118) 
                                                       at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2599) 
                                                       at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2707) 
                                                       at android.app.ActivityThread.-wrap12(ActivityThread.java) 
                                                       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1460) 
                                                       at android.os.Handler.dispatchMessage(Handler.java:102) 
                                                       at android.os.Looper.loop(Looper.java:154) 
                                                       at android.app.ActivityThread.main(ActivityThread.java:6077) 
                                                       at java.lang.reflect.Method.invoke(Native Method) 
                                                       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:865) 
                                                       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:755) 

```

This corresponds to this line in the activity's onCreate:

```
            mInputDriver = new Cap12xxInputDriver(this,
                    BoardDefaults.getI2CPort(),
                    null,
                    Cap12xx.Configuration.CAP1208,
                    keyCodes);
```

I assume it's just an invalid I2C port for the board, but I've no idea what it should be.

markmcd, 2017-06-08:
Although [over on SO](https://stackoverflow.com/questions/43700289/how-can-i-initialise-a-cap12xxinputdriver-with-androidthings) folks are saying the pico doesn't have the capacitive buttons exposed over i2c, it's over gpio. Perhaps a new sample is needed?

mangini, 2017-06-08:
That's the rainbow HAT, right? Those buttons are exposed as gpio buttons.
Use the RainbowHat driver and it should all be abstracted for you (compile
'com.google.android.things.contrib:driver-rainbowhat:0.4' on your
build.gradle)

proppy, 2017-06-08:
@markmcd in addition, you can also try the [button sample](https://github.com/androidthings/sample-button), it should work out of the box with the button `A`.

markmcd, 2017-06-19:
Superb, thank you!

closed 2017-06-19 by markmcd