androidthings/drivers-samples #14: Android Things with Raspberry Pi for Bmx280SensorDriver Example is not working

kaka120011, 2017-09-25:
It shows a exception. But I have added the permission in manifests. java.lang.RuntimeException: Unable to start activity ComponentInfo{com.kmk.test/com.kmk.test.MainActivity}: java.lang.SecurityException: Caller lacks required permission com.google.android.things.permission.MANAGE_SENSOR_DRIVERS

Fleker, 2017-09-26:
Try rebooting your device and seeing whether that works.

kaka120011, 2017-09-26:
Version 0.5 must work on version 0.5

closed 2017-09-26 by kaka120011

itamar6587, 2017-12-14:
HI, How did you solve this issue? 
I'm having the same problems..

mangini, 2017-12-14:
@itamar6587 Make sure the version of Android Things flashed on your board matches the version of the SDK used on your code and on the driver. If you are testing this sample on a board flashed with pre-0.5, the MANAGE_SENSOR_DRIVERS permissions [didn't exist at that time](https://developer.android.com/things/preview/releases.html#user_driver_permissions), and it can cause an exception similar to what you are getting.

itamar6587, 2017-12-14:
@mangini  thanks for your reply. I'll check it. thanks

Danielx303, 2018-06-13:
Also - this thing can help you - please add 2 things to your Android Manifest file: 
<uses-permission android:name="com.google.android.things.permission.MANAGE_INPUT_DRIVERS" />
<uses-permission android:name="com.google.android.things.permission.USE_PERIPHERAL_IO"/>