androidthings/drivers-samples #15: AndroidThing Pico Pro i.MX7d

aabedraba, 2018-01-02:
I'm using the AIY Assistant kit with the Pico Pro i.MX7d (instead of the Raspberry Pi.

When I compile and run I have this issue:

`01-02 13:20:32.602 2056-2056/com.example.androidthings.assistant W/main: type=1400 audit(0.0:210): avc: denied { create } for name="cgroup.procs" scontext=u:r:zygote:s0 tcontext=u:object_r:cgroup:s0 tclass=file permissive=0
01-02 13:20:32.654 2056-2056/com.example.androidthings.assistant I/zygote: Late-enabling -Xcheck:jni
01-02 13:20:33.366 2056-2056/com.example.androidthings.assistant I/InstantRun: starting instant run server: is main process
01-02 13:20:33.466 2056-2056/com.example.androidthings.assistant I/AssistantActivity: starting assistant demo
01-02 13:20:33.654 2056-2056/com.example.androidthings.assistant D/AssistantActivity: enumerating devices
01-02 13:20:33.662 2056-2056/com.example.androidthings.assistant I/AssistantActivity: initializing DAC trigger
01-02 13:20:33.697 2056-2056/com.example.androidthings.assistant E/AssistantActivity: error configuring peripherals:
                                                                                      com.google.android.things.pio.PioException: android.os.ServiceSpecificException: Unknown I/O name BCM16 (code 19)
                                                                                          at com.google.android.things.pio.GpioImpl.<init>(GpioImpl.java:53)
                                                                                          at com.google.android.things.pio.PeripheralManagerService.openGpio(PeripheralManagerService.java:170)
                                                                                          at com.google.android.things.contrib.driver.voicehat.Max98357A.<init>(Max98357A.java:59)
                                                                                          at com.google.android.things.contrib.driver.voicehat.VoiceHat.openDac(VoiceHat.java:124)
                                                                                          at com.example.androidthings.assistant.AssistantActivity.onCreate(AssistantActivity.java:310)
                                                                                          at android.app.Activity.performCreate(Activity.java:7000)
                                                                                          at android.app.Activity.performCreate(Activity.java:6991)
                                                                                          at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1214)
                                                                                          at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2731)
                                                                                          at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2856)
                                                                                          at android.app.ActivityThread.-wrap11(Unknown Source:0)
                                                                                          at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1589)
                                                                                          at android.os.Handler.dispatchMessage(Handler.java:106)
                                                                                          at android.os.Looper.loop(Looper.java:164)
                                                                                          at android.app.ActivityThread.main(ActivityThread.java:6494)
                                                                                          at java.lang.reflect.Method.invoke(Native Method)
                                                                                          at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
                                                                                          at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
                                                                                       Caused by: android.os.ServiceSpecificException: Unknown I/O name BCM16 (code 19)
                                                                                          at android.os.Parcel.readException(Parcel.java:2018)
                                                                                          at android.os.Parcel.readException(Parcel.java:1950)
                                                                                          at com.google.android.things.pio.IPeripheralManagerClient$Stub$Proxy.GetGpioClient(IPeripheralManagerClient.java:759)
                                                                                          at com.google.android.things.pio.GpioImpl.<init>(GpioImpl.java:50)
                                                                                          at com.google.android.things.pio.PeripheralManagerService.openGpio(PeripheralManagerService.java:170) 
                                                                                          at com.google.android.things.contrib.driver.voicehat.Max98357A.<init>(Max98357A.java:59) 
                                                                                          at com.google.android.things.contrib.driver.voicehat.VoiceHat.openDac(VoiceHat.java:124) 
                                                                                          at com.example.androidthings.assistant.AssistantActivity.onCreate(AssistantActivity.java:310) 
                                                                                          at android.app.Activity.performCreate(Activity.java:7000) 
                                                                                          at android.app.Activity.performCreate(Activity.java:6991) 
                                                                                          at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1214) 
                                                                                          at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2731) 
                                                                                          at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2856) 
                                                                                          at android.app.ActivityThread.-wrap11(Unknown Source:0) 
                                                                                          at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1589) 
                                                                                          at android.os.Handler.dispatchMessage(Handler.java:106) 
                                                                                          at android.os.Looper.loop(Looper.java:164) 
                                                                                          at android.app.ActivityThread.main(ActivityThread.java:6494) 
                                                                                          at java.lang.reflect.Method.invoke(Native Method) 
                                                                                          at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438) 
                                                                                          at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807) 
01-02 13:20:33.840 2056-2056/com.example.androidthings.assistant D/vndksupport: Loading /vendor/lib/hw/android.hardware.graphics.mapper@2.0-impl.so from current namespace instead of sphal namespace.
01-02 13:20:33.841 2056-2056/com.example.androidthings.assistant D/vndksupport: Loading /vendor/lib/hw/gralloc.imx7d.so from current namespace instead of sphal namespace.
01-02 13:20:33.853 2056-2056/com.example.androidthings.assistant I/display: IonManager used`

Any idea what can it be?

Fleker, 2018-01-02:
You are trying to open `BCM16`, which is not a valid pin on the imx7.d.

Looking at the device pinouts, [BCM16 on the Raspberry Pi](https://developer.android.com/things/hardware/raspberrypi-io.html) is [GPIO2_IO07](https://developer.android.com/things/hardware/imx7d-pico-io.html) on the imx7.d.

You may need to include this GPIO name as an additional parameter in the code of the AssistantActivity (Stacktrace shows `AssistantActivity:310`) to select the correct pin.

It may make more sense for this issue to go in the relevant repository.
https://github.com/androidthings/sample-googleassistant

aabedraba, 2018-01-02:
Thanks a lot @Fleker 
_Mea culpa_ for posting in the worng place.

closed 2018-01-02 by aabedraba